JdeRobot/base #1359: Cmake error

EL-SHREIF, 2019-03-25:
i have this error when I run command 

>cmake .

version:5.4.0
/home/shiko/base/src/libs/jderobotutil
/home/shiko/base/src/libs/colorspaces
/home/shiko/base/src/libs/colorspacesmm
/home/shiko/base/src/libs/easyiceconfig
/home/shiko/base/src/libs/parallelIce
/home/shiko/base/src/libs/depthLib
/home/shiko/base/src/libs/geometry
/home/shiko/base/src/libs/logger
/home/shiko/base/src/libs/ns
/home/shiko/base/src/libs/progeo
/home/shiko/base/src/libs/pioneer
/home/shiko/base/src/libs/fuzzylib
-- ALUT found at /usr/lib/i386-linux-gnu/libalut.so
-- ardronelib FOUND. Version: autonomy-gps
-- ardronelib libraries: pc_ardrone;pc_ardrone_notool;sdk;vlib;avcodec;avdevice;avfilter;avformat;avutil;swscale
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   thread
--   chrono
--   date_time
--   atomic
   -- CUDA not found!!
*** Fireware ok
*** GAZEBO LIBRARIES NOT FOUND
-- Found glog    (include: /usr/include, library: /usr/lib/i386-linux-gnu/libglog.so)
*** Goocanvasmm LIBRARIES FOUND
*** ICE ok
 kobuki_LIBRARIES 
kobuki was found in: /opt/ros/kinetic/include
kobuki LIBRARY FOUND IN /opt/ros/kinetic/lib/libkobuki.so;/opt/ros/kinetic/lib/libkobuki_dock_drive.so;/opt/ros/kinetic/lib/libecl_devices.so;/opt/ros/kinetic/lib/libecl_errors.so;/opt/ros/kinetic/lib/libecl_exceptions.so;/opt/ros/kinetic/lib/libecl_formatters.so;/opt/ros/kinetic/lib/libecl_geometry.so;kobuki_LIBRARY8-NOTFOUND;kobuki_LIBRARY9-NOTFOUND;/opt/ros/kinetic/lib/libecl_mobile_robot.so;kobuki_LIBRARY11-NOTFOUND;/opt/ros/kinetic/lib/libecl_streams.so;/opt/ros/kinetic/lib/libecl_threads.so;/opt/ros/kinetic/lib/libecl_time.so;/opt/ros/kinetic/lib/libecl_time_lite.so;/opt/ros/kinetic/lib/libecl_type_traits.so
LIBFREENECT NOT FOUND
*** libusb FOUND
Libxml++ libraries found at 
*** LIBXML2 ok
*** Nite not found
NITE2 library not found
--  NodeJS found
*** OGRe LIBRARIES FOUND: /usr/include/OGRE
-- Found OpenCV: /opt/ros/kinetic (found version "3.3.1") found components:  core flann imgproc ml features2d video objdetect 
-- Found OpenCV: /opt/ros/kinetic (found version "3.3.1") found components:  highgui calib3d 
-- Found OpenCV: /opt/ros/kinetic (found version "3.3.1") found components:  bgsegm 
***OpenCV Version: 3.3.1
***OpenCV CORE LIBS: opencv_core;opencv_flann;opencv_imgproc;opencv_ml;opencv_features2d;opencv_video;opencv_objdetect
***OpenCV GUI LIBS: opencv_highgui;opencv_calib3d
***OpenCV EXTRA LIBS: opencv_bgsegm
OpenGL libraries found at /usr/lib/i386-linux-gnu/libglut.so;/usr/lib/i386-linux-gnu/libXmu.so;/usr/lib/i386-linux-gnu/libXi.so
CMake Warning at Deps/openni/CMakeLists.txt:12 (MESSAGE):
  Defining 'i386' preprocessor macro due possible XnPlatform.h bug
Call Stack (most recent call first):
  Deps/LoadJderobotDependencies.cmake:12 (include)
  CMakeLists.txt:156 (include)

-- OpenNi found at /usr/lib/libOpenNI.so
-- OpenNI2 found at /usr/include/openni2
-- Using OpenSSL 1.0.2g
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/i386-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.
[... 169 lines cut ...]
dpkg-deb: building package 'jderobot-examples' in '.dpkg_pkgs/jderobot-examples_5.6.1_all.deb'.
libs
dpkg-deb: building package 'jderobot-libs' in '.dpkg_pkgs/jderobot-libs_5.6.1_all.deb'.
tools
dpkg-deb: building package 'jderobot-tools' in '.dpkg_pkgs/jderobot-tools_5.6.1_all.deb'.
zeroc-ice-python
dpkg-deb: building package 'jderobot-zeroc-ice-python' in '.dpkg_pkgs/jderobot-zeroc-ice-python_5.6.1_all.deb'.
jderobot
dpkg-deb: building package 'jderobot' in '.dpkg_pkgs/jderobot_5.6.1_all.deb'.
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
kobuki_LIBRARY11
    linked by target "kobuki_driver" in directory /home/shiko/base/src/drivers/kobuki_driver
kobuki_LIBRARY8
    linked by target "kobuki_driver" in directory /home/shiko/base/src/drivers/kobuki_driver
kobuki_LIBRARY9
    linked by target "kobuki_driver" in directory /home/shiko/base/src/drivers/kobuki_driver

-- Configuring incomplete, errors occurred!
See also "/home/shiko/base/CMakeFiles/CMakeOutput.log".
See also "/home/shiko/base/CMakeFiles/CMakeError.log".

=======================================================
and I make sure that I install 
Kobuki robot libraries

sleep-404, 2019-03-25:
Hello @EL-SHREIF  , just made a video on how to download and compile JDE robot in a freshly installed Ubuntu 16.04. Sorry the video is awfully long and is of less clarity, did not get enough time to edit it, as installing itself took substantial amount of time. You can skip many parts as most of the video though. I apologise once again.
There are a few things you need to know before watching the video:
1.) I installed Opencv in the video , which was not required for 16.04(required only for 18.04)
2.) After installing dependencies ,while building, errors popped up regarding **opencv** and **roscpp** , all I did was closed all the terminal and re-opened them. This fixed it . You need not install anything extra
3.) For errors on zero-ice-python , please refer to [this](https://github.com/JdeRobot/base/issues/1345) issue
4.) Error regarding **OpenCV_INCLUDE_DIR** was fixed by removing the build directory and again creating it back.
The video can be found [here](https://www.youtube.com/watch?v=vSjTg_Nfo_8&feature=youtu.be). Hope you find this helpful.

Pravin854, 2019-03-25:
Try `sudo apt-get install ros-kinetic-kobuki`

EL-SHREIF, 2019-03-25:
I have this error @sleep-404 
 
>shiko@PC:~/base$ sudo apt-get install jderobot-deps-dev

[sudo] password for shiko: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 jderobot-deps-dev : Depends: jderobot-openni2 but it is not installable
                     Depends: jderobot-libfreenect-dev but it is not installable
                     Depends: qfi but it is not installable
                     Depends: ardronelib but it is not installable
                     Depends: libzeroc-ice3.6 but it is not installable
                     Depends: zeroc-ice-utils but it is not installable
                     Depends: libzeroc-icestorm3.6 but it is not installable
                     Depends: jderobot-openni2 but it is not installable
                     Depends: jderobot-libfreenect-dev but it is not installable
                     Depends: qfi but it is not installable
E: Unable to correct problems, you have held broken packages.

EL-SHREIF, 2019-03-26:
> I have this error @sleep-404
>
>
> [sudo] password for shiko:
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Some packages could not be installed. This may mean [...]

this can be solve by using  sudo aptitude not sudo apt-get

EL-SHREIF, 2019-03-26:
> i have this error when I run command
>
>
> version:5.4.0
> /home/shiko/base/src/libs/jderobotutil
> /home/shiko/base/src/libs/colorspaces
> /home/shiko/base/src/libs/colorspacesmm [...]

this can be solve by updating the repo by the existing one

closed 2019-03-26 by EL-SHREIF