JdeRobot/base #1362: "JdeRobot" installation instructions in the base repo

jmplaza, 2019-03-29:
The latest installation instructions for "JdeRobot" should be in this repo, maybe as a README.md file. This should be the reference place for them, instead of the wiki.

In the same way, "JdeRobot" name will now refer to the organization itself and "base" for the software we named "JdeRobot" so far. This change follows the movement in our organization to avoid a monolithic repository (as some years ago) and to work instead with distributed repositories: one for VisualStates tool, another for assets, other for visualization tools, etc.. This also shows the migration of JdeRobot from a framework to a toolkit, which mostly rely on ROS drivers.

"JdeRobot/base" stills contains a collection of drivers, but they will be no longer maintained as long as a ROS equivalent exists.

PaulNicolasHunter, 2019-03-29:
@jmplaza do you find any work around to install it? [this issue](https://github.com/JdeRobot/base/issues/1350) doesn't seem to address the solution properly, I can't even find the binary release in packages.ubuntu?

can you share the link of wiki?

sleep-404, 2019-04-02:
Here is a link to updated video tutorial as well.
[Compiling JDE/base from source](https://youtu.be/uzC48QFVRAM)
Hope that helps the community

sleep-404, 2019-04-02:
Here is an updated video link that might help new users with their installations:
[Compiling JDE/base from source (Updated)
](https://www.youtube.com/watch?v=DslZX07qeeE)

commit 30301858 (2019-04-03): Merge pull request #1365 from fqez/update_README

updated README with installation instructions fix #1362
closed 2019-04-03 by fqez