JdeRobot/base #1368: Compile JdeRobot ubuntu 18.04

talregev, 2019-04-09:
I compile the base a part of the GSoc challenge:
https://jderobot.org/store/jmplaza/uploads/gsoc/gsoc2019-installation_test.pdf
There is a small typo in this docs.
The link instructions of how to compile the code is not valid anymore. I guess you transfer the instructions in the github itself. 

I read this question forum:
https://developers.jderobot.org/t/installation-of-academy-on-ubuntu-18-04/100
And this issue: 
https://github.com/JdeRobot/base/issues/1343.

From the forum:   
> the support for TurtleBot-2 robot in ROS Melodic is not fine yet AFAIK    
I was able to compile and run turtlebot 2 kobuki in ros melodic and run it on gazebo 9.7   

I didn't see there is others branch of ubuntu 18.04, so for compile the code I made the changes  
myself, so I want to add it to pull request if you create another branch for it.    
https://github.com/talregev/base 
After looking the other branches, I think my code is more understandable  and handle the close camera. 

Please add new branch from the master for ubuntu 18.04 and I will send a pull request.

fqez, 2019-04-10:
Hi @talregev 

thank you so much for you effort on this matter, it's indeed needed work on this issue. I checked out your last commit in your fork, but I am affraid that those changes were already made in this PR #1363. I don't know if you add more changes to this matter because but please, feel free to compare your branch with [this one](https://github.com/JdeRobot/base/tree/ubuntu_bionic) where those changes were made.

If you have more changes to add to that branch, feel free to create a new pull request adding all the changes you want/need se we will be glad to test them and merge them if they are correct.

Regards!

talregev, 2019-04-10:
Thank you for your respond. 
I make pr to [ubuntu_bionic][1].

not sure about 2 things:
- [f1/motor.cc][1] line 114.   In the master it like as i commit. [f1/motor.cc][2]
- [f1/pose3d.cc][3] line 92.  In the master it like as i commit. [f1/pose3d.cc][4]

[1]:https://github.com/JdeRobot/base/compare/ubuntu_bionic...talregev:ubuntu_bionic?expand=1#diff-1fb944150847b4a171ad4a2bfc9d32eaR114
[2]:https://github.com/JdeRobot/base/blob/master/src/drivers/gazebo/plugins/f1/motors.cc#L115
[3]:https://github.com/JdeRobot/base/compare/ubuntu_bionic...talregev:ubuntu_bionic?expand=1#diff-a3f8cd65e6f17b89eaa97321739a3b21R92
[4]:https://github.com/JdeRobot/base/blob/master/src/drivers/gazebo/plugins/f1/pose3d.cc#L92

Also I didn't know if the close camera it good behavior. In the branch it comment all close camera, and in my branch I was reset the share pointer (this action close the connection).
But I leave the behavior of close camera,  as it is in the branch.

[1]:https://github.com/JdeRobot/base/tree/ubuntu_bionic

closed 2019-05-23 by fqez