JdeRobot/base #1381: [Tools] Move Colortuner to its own repository

fqez, 2019-08-01:
In order to implement an easy to maintain CI/CD system for our tools, it's better to move each tool to its own repository and make them atomic. This way every tool is standalone and base independent. Moving ColorTuner to https://github.com/JdeRobot/ColorTuner.git

pankhurivanjani, 2019-08-03:
Hi!
Before refactoring the Colortuner for ROS1 I am trying to run it with ICE drivers only.  I am able to get the camera images from camserver but running 

```$ colorTuner /opt/jderobot/share/jderobot/conf/colorTuner_py.yml ```

is giving the errors :

``` Traceback (most recent call last):
  File "/opt/jderobot/share/jderobot/python/colorTuner_py/colorTuner.py", line 25, in <module>
    import comm
  File "/opt/jderobot/lib/python2.7/comm/__init__.py", line 1, in <module>
    from .communicator import Communicator
  File "/opt/jderobot/lib/python2.7/comm/communicator.py", line 4, in <module>
    from .laserClient import getLaserClient
  File "/opt/jderobot/lib/python2.7/comm/laserClient.py", line 4, in <module>
    from .ice.laserIceClient import LaserIceClient
  File "/opt/jderobot/lib/python2.7/comm/ice/laserIceClient.py", line 21, in <module>
    import jderobot
  File "/opt/jderobot/lib/python2.7/jderobot/__init__.py", line 8, in <module>
    import ardroneextra_ice
  File "/opt/jderobot/lib/python2.7/ardroneextra_ice.py", line 29, in <module>
    _M_jderobot._t_ArDroneExtra = IcePy.defineValue('::jderobot::ArDroneExtra', Ice.Value, -1, (), False, True, None, ())
AttributeError: 'module' object has no attribute 'defineValue' ```

Is it some environment sourcing issue or anything else in documentation to run it?

fqez, 2019-08-03:
Hi @pankhurivanjani I've just tested it and it's working. Have you sourced both JdeRobot and ROS?
```
source /opt/jderobot/setup.bash
source /opt/ros/melodic/setup.bash
```

Regards.

pankhurivanjani, 2019-08-03:
I had sourced the path but it was version error in ICE, corrected that one. 

I stumbled upon another error/warning next

``` warnings.warn(message, YAMLLoadWarning, stacklevel=3)
yaml.YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details. ```

This is due to [PyYAML yaml.load(input) Deprecation](https://github.com/yaml/pyyaml/wiki/PyYAML-yaml.load(input)-Deprecation) as explained in the link. I changed the  file mentioned below to correct it.

'/opt/jderobot/lib/python2.7/config/config.py '

line 67 to  ``` cfg=yaml.load(stream,Loader=yaml.FullLoader)  ``` 

But I am still getting errors:

``` $ colorTuner /opt/jderobot/share/jderobot/conf/colorTuner_py.yml 
loading Config file /opt/jderobot/share/jderobot/conf/colorTuner_py.yml
Receiving ColorTuner.Camera Image from ICE interfaces
Traceback (most recent call last):
  File "/opt/jderobot/lib/python2.7/comm/ice/cameraIceClient.py", line 53, in __init__
    basecamera = ic.stringToProxy(proxyStr)
AttributeError: 'NoneType' object has no attribute 'stringToProxy' ```

This can be a dependency issue. 

Also, you tested the tool on ROS Kinetic or Melodic?

fqez, 2019-08-03:
Hi, about the error change the colorTuner_py.yml in `/opt/jderobot/share/jderobot/config` the server property from `Ice` to `1` and it should work.

pankhurivanjani, 2019-08-03:
yeah, this was the error. Got it worked. I will document these possible error sources in ReadMe. 
Thank you!

closed 2019-08-07 by fqez