JdeRobot/base #1386: [Question] GSoC Queries

Korusuke, 2020-02-15:
Are the Programming Tests listed going to be updated, as currently, 2019 ones are linked?

Also, I guess the Programming Tests Table is not updated yet.

jmplaza, 2020-02-15:
Hi @Korusuke , 

you are right, the Programming Tests Table has not been updated yet. 

Yes, the [JdeRobot mailing list](https://gsyc.urjc.es/cgi-bin/mailman/listinfo/jde-developers) is the preferred communication way for non-code questions. Don't feel ashamed, just subscribe yourself and use it.

closed 2020-02-15 by jmplaza

Korusuke, 2020-02-22:
Sorry for continuing this thread. 
As advised I have mailed on the mailing list but I doubt that anyone has received it since the view count [here](http://jderobot-developer-list.2315034.n4.nabble.com/) is stuck on 2 from the day I sent it.

> Yes, the [JdeRobot mailing list](https://gsyc.urjc.es/cgi-bin/mailman/listinfo/jde-developers) 

As already mentioned in my mail, the archives linked at the above link doesn't work for me.
Am I am doing something wrong here?

Should I directly contact the mentors for discussing the project or wait for the challenges to be updated?

jmplaza, 2020-02-23:
Be patient :-) , I've just answered in the mailing list. Cheers!