JdeRobot/base #1387: The Repo needs upgdation to Python 3

ayushmankumar7, 2020-02-22:


jmplaza, 2020-02-23:
Hi @ayushmankumar7, you are right. Regarding Python3, we are pretty conditioned by ROS. Now or reference ROS distribution is ROS-Melodic, which still only supports Python 2.7. Things will change on next May, as [OpenRobotics will launch ROS-Noetic distribution](https://discourse.ros.org/t/planning-future-ros-1-distribution-s/6538), which will support Python 3.

So let's wait to ROS-Noetic for migrating JdeRobot-base.

Meanwhile we are now focused on repository cleaning and reorganization (@fqez). For instance, all the tools (ColorTuner...) will be removed soon as they have now their corresponding repositories. This was the reference repo when JdeRobot was a framework, and so, its had many things inside, too many. Now JdeRobot is a toolkit, spread in several repositories. In addition, [ICE communication middleware](https://zeroc.com/products/ice) was the choice that time but we switched to use ROS middleware three years ago.

From now on JdeRobot-base is the repository for ROS-based drivers, both for real robots or simulated robots, which are not in the official ROS packages. This way [JdeRobot Academy](https://jderobot.github.io/RoboticsAcademy/) may use those robots, for instance.

ayushmankumar7, 2020-02-23:
Thanks for the info. :)

closed 2021-02-21 by jmplaza