JdeRobot/base #1388: E: Unable to locate package jderobot

ayushmankumar7, 2020-02-23:
I faced many problems while setting up the application on my local machine.

[image]

jmplaza, 2020-02-23:
Hi @ayushmankumar7 , take a look at the [installation instructions for JdeRobot-base](https://jderobot.github.io/RoboticsAcademy/installation/#jderobot-base) in JdeRobot Academy. They are the most updated ones.

The README file at JdeRobot-base should be updated with that content too...

ayushmankumar7, 2020-02-23:
Release file for "eoan" not present. 

[image]

Following Error is popping up. 
[image]

I tried doing installing it using the Bionic dist on my eoan dist. It says "you have held broken packages"

jmplaza, 2020-02-24:
Hi @ayushmankumar7, Ubuntu 19.04 (Eoan Ermine) is NOT supported, Jderobot-base debian package will NOT be created for it. We only support LTS Ubuntu distributions.

Alternatively you may work (a) directly from JdeRobot-base source code or (b) use a virtual machine or docker container with Ubuntu 18.04 and install there the jderobot-base debian package.

closed 2020-02-24 by jmplaza

ayushmankumar7, 2020-02-24:
It's alright sir. Thanks you the information.