JdeRobot/base #1390: Received the following error while compiling according to the given instructions

niveditarufus, 2020-02-28:
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv 24E521A4
Executing: /tmp/apt-key-gpghome.FJ8lZff9Fa/gpg.1.sh --keyserver keyserver.ubuntu.com --recv 24E521A4
gpg: keyserver receive failed: No keyserver available

khmariem, 2020-03-07:
There is a possibility that a firewall is blocking the reception on the hkp port, because keyserver.ubuntu.com is actually hkp://keyserver.ubuntu.com not http://keyserver.ubuntu.com

So, I suggest you force the reception on the http port. Try this, it might solve your issue for now:
`sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv 24E521A4`

fqez, 2020-03-10:
Hi! did you manage to make this work with the suggestion from @khmariem??

sasilva1998, 2020-03-18:
Hi, I am having the same issue. I was following the installation from https://jderobot.github.io/RoboticsAcademy/installation/

Tried @khmariem suggestion and got the following
```bash
Executing: /tmp/apt-key-gpghome.JzRtIWl45P/gpg.1.sh --keyserver hkp: //keyserver.ubuntu.com: 80 --recv 24E521A4
gpg: Key F4F3860524E521A4: "Francisco Perez <f.perez475@gmail.com>" not changed
gpg: total number of keys processed: 1
gpg: unchanged: 1
```
Still haven't managed to run `sudo apt install jderobot` successfully.

sasilva1998, 2020-03-18:
It look it is a problem with libraries and that I should install ardronelib however I still can't install it either, I have a kind of version problemas with all these libraries.

sasilva1998, 2020-03-18:
Yes, basically the problem was this:

```
The following packages have unfulfilled dependencies:
  libpulse-dev: Depends on: libpulse0 (= 1: 11.1-1ubuntu7.4) but 1: 11.1-1ubuntu7.5 should be installed
                 Depends on: libpulse-mainloop-glib0 (= 1: 11.1-1ubuntu7.4) but 1: 11.1-1ubuntu7.5 should be installed
E: Problems cannot be corrected, you have defective packages withheld.
```

It was a version problem. I understood that when trying to install jderobot, ardronelib was a dependency to be installed, and couldn't because of these version of `libpulse0` and `libpulse-mainloop-glib0`. So solved it using aptitude and downgrading the version of all the needed components to version 7.4.

niveditarufus, 2020-03-18:
@khmariem @fqez I am sorry to have replied so late.... yes the suggestion was helpful and I was able to get to install it.

closed 2021-02-21 by jmplaza