JdeRobot/base #1391: Error while installing dependencies

bhatakti-atama, 2020-02-29:
Hi.
I am trying to install dependencies for building JdeRobot but I am getting the following error message
[image]

khmariem, 2020-03-07:
I believe this can be an issue with packages in your own computer from the first place. To be sure, can you send the output of the following command? `sudo apt-get upgrade`

bhatakti-atama, 2020-03-08:
[image]

ojgarciab, 2020-03-09:
Are you using **bionic** (Ubuntu 18.04 LTS) distro? The package [is available](https://packages.ubuntu.com/bionic/libgstreamer1.0-dev).

ojgarciab, 2020-03-09:
The package was upgraded to [version 1.0 in **bionic** (Ubuntu 18.04 LTS)](https://packages.ubuntu.com/bionic/libgstreamer1.0-dev) so that dependencies [are por **xenial** (Ubuntu 16.04 LTS)](https://packages.ubuntu.com/xenial/libgstreamer0.10-dev).

Could you share jderobot your repository file?

ojgarciab, 2020-03-10:
As a workaround (because `jderobot-deps-dev` is a metapackage, with an empty `data` package and a `control` file with the same info than `Packages` file), you could install packages manually:

    apt install libeigen3-dev libdc1394-22-dev libgazebo9-dev libgsl-dev libgtk2.0-bin \
      libgtk2.0-cil libgtk2.0-common libgtk2.0-dev libgtkgl2.0-dev libgtkglext1-doc libgtkglext1-dev \
      libgtkmm-2.4-dev libglademm-2.4-dev libgnomecanvas2-dev libgnomecanvasmm-2.6-dev \
      libgtkglextmm-x11-1.2-dev libgoocanvasmm-2.0-dev zeroc-ice-slice libzeroc-ice-dev \
      libusb-1.0-0-dev libxml++2.6-dev libogre-1.9-dev libopencv-dev freeglut3-dev libglu1-mesa-dev \
      libpcl-dev python-numpy python3-numpy python-pip python3-pip python-pyqt5 \
      python-pyqt5.qtsvg python3-pyqt5 python3-pyqt5.qtsvg python-pyqt5.qsci python-yaml qfi \
      libqt5opengl5-dev libqt5svg5-dev qtbase5-dev libtinyxml-dev libyaml-cpp-dev git cmake make \
      gcc-5 g++-5 gcc g++ pkg-config libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev \
      ardronelib libboost-atomic1.65.1 libboost-chrono1.65.1 libboost-date-time1.65.1 \
      libboost-filesystem1.65.1 libboost-system1.65.1 libboost-thread1.65.1 libdc1394-22 gazebo9 \
      libgoogle-glog-dev libgsl23 gsl-bin libgtk2.0-0 libgtkgl2.0-1 libgtkmm-2.4-1v5 libgtkglext1 \
      libglademm-2.4-1v5 libgnomecanvas2-0 libgnomecanvasmm-2.6-1v5 \
      libgtkglextmm-x11-1.2-0v5 libgtkglext1 libgoocanvasmm-2.0-6 libzeroc-ice3.7 zeroc-ice-utils \
      libzeroc-icestorm3.7 libusb-1.0-0 libxml++2.6-2v5 libxml2 libogre-1.9.0v5 libopencv-dev \
      freeglut3 libglu1-mesa libpcl-common1.8 libpcl-octree1.8 libpcl-io1.8 libpcl-kdtree1.8 \
      libpcl-search1.8 libpcl-sample-consensus1.8 libpcl-filters1.8 libpcl-features1.8 libpcl-ml1.8 \
      libpcl-segmentation1.8 libpcl-visualization1.8 libpcl-surface1.8 libpcl-registration1.8 \
      libpcl-keypoints1.8 libpcl-tracking1.8 libpcl-recognition1.8 libpcl-stereo1.8 libpcl-apps1.8 \
      libpcl-outofcore1.8 libpcl-people1.8 libpthread-stubs0-dev libbz2-dev libssl-dev python-numpy \
      python-pip python-pyqt5 python-pyqt5.qtsvg python-pyqt5.qsci python-yaml qfi libqt5core5a \
      libqt5opengl5 libqt5gui5 libqt5svg5 libqt5widgets5 ros-melodic-roscpp ros-melodic-std-msgs \
      ros-melodic-cv-bridge ros-melodic-image-transport ros-melodic-roscpp-core ros-melodic-rospy \
      ros-melodic-nav-msgs ros-melodic-geometry-msgs ros-melodic-mavros \
      ros-melodic-kobuki-driver ros-melodic-kobuki-core ros-melodic-turtlebot3-gazebo \
      ros-melodic-rplidar-ros ros-melodic-urg-node libtinyxml2.6.2v5 libxerces-c3.2 libyaml-cpp0.5v5

jmplaza, 2020-03-10:
Hi @bhatakti-atama , 

watch out that ONLY Ubuntu-18.04, ROS Melodic and Gazebo-9 are supported right now. It is strongly recommended to install JdeRobot/base from debian package following [this recipe](https://jderobot.github.io/RoboticsAcademy/installation/#jderobot-base). There is no need to install jderobot-deps-dev on your own, just follow that recipe and it should work.

In other Ubuntu distributions (a) you could work with installation from source code, and that is complex as JdeRobot/base depends on many third party software pieces. We created the debian package for JdeRobot/base to simplify its installation, it is available and ready to use. Alternatively (b) you could use virtual machines or docker containers with Ubuntu 18.04.

fqez, 2020-03-10:
Hi all! I'm going to release a new version of this repository with all the dependencies fixed, so you can install it through APT. The new release will be ready this week hopefully (dependin on my free time) so if you can wait, I will let you know when the new packages are ready.

Meanwhile, if you don't want to wait for the release, you can install all the dependencies needed from the `scripts/install_deps.sh` script, which contains the dependencies updated for Ubuntu Bionic (18.04).

Nonetheless, I strongly recommend you to wait for the release, just to be sure you don't break your APT.

Sorry for all the inconveniences. Stay tuned!

closed 2021-02-21 by jmplaza