JdeRobot/base #1402: jderobot-base installation

manoharbhat, 2020-11-09:
Hello I was trying to install jderobot from the jderobot.github.io installation instructions, I have followed all the steps, and when this command was executed - **sudo apt install jderobot** - I got the following error 

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 jderobot : Depends: ardronelib but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

pls, let me know how to install it correctly. currently, I am using Ubuntu 18.04 LTS and ROS melodic

jmplaza, 2020-11-09:
Hi @manoharbhat,
jderobot-base itself is deprecated. If you want to use one of our recent tools RoboticsAcademy, VisualCircuit or BehaviorStudio... they no longer depend on jderobot-base. Old tools have been mostly deprecated too, as they are no longer maintained. What JdeRobot tool do you want to use?

manoharbhat, 2020-11-09:
Robotics Academy

jmplaza, 2020-11-10:
Hi @manoharbhat , I suggest to start your experience with RoboticsAcademy with these two exercises using their web templates: 

1. [Visual Follow Line](https://jderobot.github.io/RoboticsAcademy/exercises/AutonomousCars/follow_line/)
2. [Local navigation with VFF](https://jderobot.github.io/RoboticsAcademy/exercises/AutonomousCars/obstacle_avoidance)

Their backend are not installed locally, but they run inside a [docker container](https://hub.docker.com/r/jderobot/robotics-academy) already ready to use. Just follow the corresponding installation instructions (for web templates) on each exercise page and feel free to report any problem in the RoboticsAcademy repository.

They both are the exercises already in the brand new 2.3 release of RoboticsAcademy. The remainder are still in 2.1 release (ROS node templates) or even in 2.0 (ROS node templates and jderobot-base + jderobot-assets as dependence).

closed 2020-11-10 by jmplaza